rosenkolev/home-assistant-gpio-integration #8: RGB LED - Error "object has no attribute '_blink_off'"

TCWORLD, 2025-04-04:
I've just been trying to set up this integration on my Pi.

When trying to use the RGB Light integration, whenever I try to change anything, I get an error message pop up:

[image]

In the logs I see the following error:

```
[548099701008] Unexpected exception

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 648, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1703, in async_turn_on
    await self.hass.async_add_executor_job(ft.partial(self.turn_on, **kwargs))
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/gpio_integration/light.py", line 225, in turn_on
    self._blink_off()
    ^^^^^^^^^^^^^^^
AttributeError: 'RgbGpioLight' object has no attribute '_blink_off'

```

I'm using the latest version of Home Assistant, and installed this add-on (v2.0.3) using HACS.

---

In case it makes a difference, I'm using supervised mode on Raspbian (technically unsupported by HA folk, but seems to work fine).

TCWORLD, 2025-04-04:
Controlling the LED using just the "LED (PWM)" variation works correctly.

TCWORLD, 2025-04-04:
I assume that line should just be `self._blink("off", pwm=True)` instead of `self._blink_off()`. Seems to fix the issue assuming that's the behaviour you're intending. That then yeilds an issue with async writes, specifically a call to `self.async_write_ha_state()`, which I assume based on the regular PWM light code should be `self.schedule_update_ha_state()`.

rosenkolev, 2025-04-05:
I will work on this today and test the other issue you submitted.

closed 2025-04-05 by rosenkolev