rosenkolev/home-assistant-gpio-integration #17: Issue with DHT22

small-solar-diy, 2025-04-11:
Hi,

I have it working on my Raspberry PI 4 in HA operating system native.

I have a switch that toggles a led in GPIO23, that works from the dashboard.

I was trying to get the DHT22 (but I think it's a DHT11, same thing???) to work giving an error. Is the DHT22 different from 11 from a protocol point of view?

[image]

The error:
  Needs attention
  temp
  No devices or entities
  Failed to set up:
  Check the logs
  
The first part of the log.

2025-04-11 02:32:32.580 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Chambre for gpio_integration
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/native.py", line 237, in export
    result = self._exports[pin]
             ~~~~~~~~~~~~~^^^^^
KeyError: 23

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/native.py", line 247, in export
    result = os.open(self.path_value(pin),
                     os.O_RDONLY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/gpio/gpio23/value'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/gpio_integration/__init__.py", line 44, in async_setup_entry
    hub = Hub(entry.data)
  File "/config/custom_components/gpio_integration/hub.py", line 55, in __init__
    self.controller = DHT22Controller(self.config)
                      ~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/config/custom_components/gpio_integration/controllers/sensor.py", line 61, in __init__
    self._io = DHT22(config.pin)
               ~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
  File "/config/custom_components/gpio_integration/_devices.py", line 258, in __init__
    super().__init__(
    ~~~~~~~~~~~~~~~~^
        pin,
        ^^^^
        bounce_time=0.000_005,
        ^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/gpio_integration/_devices.py", line 194, in __init__
    self.pin.edges = "both"
    ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/__init__.py", line 441, in <lambda>
    lambda self, value: self._set_edges(value),
                        ~~~~~~~~~~~~~~~^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/native.py", line 519, in _set_edges
    self.factory.fs.export(self._number)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/native.py", line 251, in export
    with io.open(self.path('export'), 'wb') as f:
         ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument

What am I doing wrong?

Thanks!

small-solar-diy, 2025-04-11:
> It seems it's not related to the DHT sensor, but errors when trying to open the pin https://github.com/gpiozero/gpiozero/blob/ceaec6150c71654585113320113cb45f0aa784b5/gpiozero/pins/native.py#L234
>
> Not sure what is the reason but you may try:
>
>   *   Restart home assistant fully so if some other process is using the pin to release it
>   *   try another pin
>   *   try to use pigpio instead of the native factory
>
> Looking into DHT11 and DHT22 datasheet it seam the data protocols are exactly the same, so it shoud work.
>
>
>
> It seems it's not related to the DHT sensor, but errors when trying to open the pin https://github.com/gpiozero/gpiozero/blob/ceaec6150c71654585113320113cb45f0aa784b5/gpiozero/pins/native.py#L234
>
> Not sure what is the reason but you may try:
>
>   *   Restart home assistant fully so if some other process is using the pin to release it
>   *   try another pin
>   *   try to use pigpio instead of the native factory
>
> Looking into DHT11 and DHT22 datasheet it seam the data protocols are exactly the same, so it shoud work.

Thanks, will try a few things with docker privileges and let you know what works.

rosenkolev, 2025-04-17:
> Thanks, will try a few things with docker privileges and let you know what works.
> […](#)

@small-solar-diy Any update on this?

ibrahimwaly, 2025-06-09:
Hi,
    I've got this error 
Failed to setup:
Check the logs

Is there any modification required in configuration.yaml ?

____________________________________
Error setting up entry dht11 for gpio_integration
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/gpio_integration/__init__.py", line 44, in async_setup_entry
    hub = Hub(entry.data)
  File "/config/custom_components/gpio_integration/hub.py", line 56, in __init__
    self.controller = DHT22Controller(self.config)
                      ~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/config/custom_components/gpio_integration/controllers/sensor.py", line 65, in __init__
    self._io = DHT22(config.pin)
               ~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
  File "/config/custom_components/gpio_integration/_devices.py", line 297, in __init__
    super().__init__(
    ~~~~~~~~~~~~~~~~^
        pin,
        ^^^^
        bounce_time=0.000_005,
        ^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/gpio_integration/_devices.py", line 227, in __init__
    pin_factory=get_pin_factory(),
                ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/gpio_integration/_pin_factory.py", line 75, in get_pin_factory
    pin_factory = create_pin_factory(name, pin_factory_class)
  File "/config/custom_components/gpio_integration/_pin_factory.py", line 50, in create_pin_factory
    return pin_factory_class()
  File "/usr/local/lib/python3.13/site-packages/gpiozero/pins/pigpio.py", line 93, in __init__
    raise IOError(f'failed to connect to {host}:{port}')
OSError: failed to connect to localhost:8888

ibrahimwaly, 2025-06-14:
Yes, I've installed pigpiod on RPI5 which was HA host with no hope, I realized after that it won't work with RPI5 anyway, so I tried again with RPI 4 and I've succeeded to add the integration and it shows 1 device 2 entities but the sensor DHT11 gives temp 0.0, Humidity 0.0, I've replaced it with DHT22 the same thing

ibrahimwaly, 2025-06-16:
2025-06-16 13:28:08.980 WARNING (Thread-2) [custom_components.gpio_integration.core] GPIO11 (DHT22): invalid start bits
2025-06-16 13:28:49.120 WARNING (Thread-2) [custom_components.gpio_integration.core] GPIO11 (DHT22): invalid start bits
2025-06-16 13:29:09.191 WARNING (Thread-2) [custom_components.gpio_integration.core] GPIO11 (DHT22): invalid start bits
2025-06-16 13:29:21.639 DEBUG (MainThread) [custom_components.gpio_integration.core] GPIO11 (DHT22) (DHT22Controller): releasing
2025-06-16 13:29:21.765 DEBUG (MainThread) [custom_components.gpio_integration.core] None (DHT22Controller): auto read loop stopped
2025-06-16 13:29:21.766 DEBUG (MainThread) [custom_components.gpio_integration.core] Hub: type "SENSOR_DHT22"
2025-06-16 13:29:21.772 DEBUG (MainThread) [custom_components.gpio_integration.core] GPIO11 (DHT22) (DHT22Controller): auto read loop started

ibrahimwaly, 2025-06-16:
> 2025-06-16 13:28:08.980 WARNING (Thread-2) [custom_components.gpio_integration.core] GPIO11 (DHT22): invalid start bits 2025-06-16 13:28:49.120 WARNING (Thread-2) [...]

I've found values coming up from DHT11 but it's wrong ( Humidity 1,331.4%, Temp 640.5 C)