rosenkolev/home-assistant-gpio-integration #18: Use to control motor?

jatgm1, 2025-04-21:
Is it possible to use this to control the pins on a raspberry pi when the pi is connected to a yahboom tank kit, so that I could install home assistant onto this raspberry pi, capture video from the camera, and make a dashboard that can both move the vehicle and display the video, then integrate it into another home assistant instance so I can try to automate its controls? I think I'll have to install a supervised home assistant instance on to raspbian or something so I can also run a video server to broadcast the video and then put that video into a dashboard on home assistant, using buttons to control the gpio pins which use pwm to control the tank? Maybe make it listen for mqtt events and perform actions when their received to make remotely controlling it easier.

rosenkolev, 2025-04-22:
Hi,
What you try to do sound complex and I am not sure how you can achieve all this with Home Assistant.
I guess you can make some automation that may work.

About controlling the motor you can use switch entity and relay:
If you use 2 motors and 2 relays and the Robot will go only forward (I asume the motors run on 12V):
```mermaid
---
title: GPIO with 2 relays and 2 motors
---
flowchart TB
  subgraph Raspberry GPIO
    A[GPIO 23]
    B[GPIO 25]
  end

  subgraph Relay1
    I1[In]
    C1[Control]
    O1[Out]
  end
  subgraph Relay2
    I2[In]
    C2[Control]
    O2[Out]
  end
  PWR([+12V])
  G([GND])
  M1((Left Motor))
  M2((Right Motor))
  PWR --- I1
  PWR --- I2
  A --> C1
  B --> C2
  O1 --> M1
  O2 --> M2
  G --> M1
  G --> M2
```

With this setup you will be able to move your Robot car when switching the pins 23 and 25 on and off. You can do this by an automation.

If you want to be able to do a reverse on the motors you better get and Motor Control Board like shown in [Build a Buggy](https://projects.raspberrypi.org/en/projects/build-a-buggy) project. There are several options out there. Then you can have 4 switches:
* Switch 1 - Left Motor forward
* Switch 2 - Left Motor backward
* Switch 3 - Right Motor forward
* Switch 4 - Right Motor backward

But this is just the motion of your robot, Now you have to write some automations and scripts to make your robot move as expected and I can not help you with that.

rosenkolev, 2025-04-23:
Now I see You are refering to a specific hardware [yahboom tank kit](https://category.yahboom.net/products/g1tank). Am I correct? If this is the case I see that the manifacturer is providing it's own [Software and OS](http://www.yahboom.net/study/G1-T-PI). I don't think you need Home Assistant to program it. I suggest looking at the manifacturer documentation for answers.

jatgm1, 2025-05-10:
> Now I see You are refering to a specific hardware [yahboom tank kit](https://category.yahboom.net/products/g1tank). Am I correct? If this is the case I see that the manifacturer is providing it's [...]

you are absolutely right that it does come with software, however it does not work very well or allow me to integrate it with a home assistant. so i thought. ill just make it into a home assistant and use the remote home assistant to allow me to access a dashboard on the device to do all the actions i need. which is just moving forward and backward and streaming video. i removed most of the sensors because they seemed excessive to me. i just need it for remotely checking on things. and i need it to feed the video into blue iris so its captured to my security system and the backed up.

jatgm1, 2025-05-10:
> Now I see You are refering to a specific hardware [yahboom tank kit](https://category.yahboom.net/products/g1tank). Am I correct? If this is the case I see that the manifacturer is providing it's [...]

I just installed home assistant to it today since I had the time this morning. I found documentation for using the pins for a sensor or for a switch. Is there a way to control a servo motor with it or use pwm for speed for the switches? I just didn't see any documentation for this part specifically. I asked the manufacturer for information related to which pins it used for the servo motors and drive motors. I'll see what happens.

rosenkolev, 2025-05-11:
The extension don't support Servo right now but I have created a Servo initial code. And will create a Servo enhancement.
https://github.com/rosenkolev/home-assistant-gpio-integration/pull/19

jatgm1, 2025-05-11:
:D I really wanna make a robot powered by home assistant. I can do so much with it... Using the gpio pins is really useful for that. Too bad there isn't a ros2 addon for home assistant. It would be perfect. I really don't understand why gpio pins aren't a core integration. I mean, It's the most DIY way to run things that works consistently and would be compatible. Heck the home assistant yellow was literally made for the pi I have no idea why you wouldn't lean in to it more.

jatgm1, 2025-05-11:
Also, how would I use a number helper set to 1-100 as a variable and use that to control a pin using pwm, like for a motor or fan? There are two relays for each motor for forward and reverse (h-bridge?) and I want to be able to adjust the number variable slider to determine the speed at which each motor turns, I believe it's using pin 13 and 16 for the speed and four other pins I already set up as forward and reverse for each respective motor. I just need to figure out how to make it control a pwm for speed adjustment.

jatgm1, 2025-05-12:
How do I make it use a servo? Also the fan speed worked fantastic

jatgm1, 2025-05-12:
It doesn't appear when I click add, no servo option. It's just "gpio integration '

jatgm1, 2025-05-12:
[image]

rosenkolev, 2025-05-12:
I have just released the Servo support with version 2.1.0 so it will take time to show for you

jatgm1, 2025-05-12:
Oh okay Sorry.

jatgm1, 2025-05-12:
So I just tried it and it refuses to load the other ones I already have. Should I redownload the older version that loads and works, delete the entities and the the integration, then restart and install the new one?

jatgm1, 2025-05-12:
I'll post the logs

Logger: homeassistant.setup
Source: setup.py:344
First occurred: 12:43:22 PM (1 occurrence)
Last logged: 12:43:22 PM

Setup failed for custom integration 'gpio_integration': Unable to import component: Exception importing custom_components.gpio_integration
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/gpio_integration/__init__.py", line 10, in <module>
    from .hub import Hub
  File "/config/custom_components/gpio_integration/hub.py", line 3, in <module>
    from .config_flow import fill_schema_missing_values
  File "/config/custom_components/gpio_integration/config_flow.py", line 42, in <module>
    from .schemas.servo import SERVO_SCHEMA, validate_servo_data
  File "/config/custom_components/gpio_integration/schemas/servo.py", line 23, in <module>
    vol.Optional(CONF_DEFAULT_STATE, default=0): cv.int,
                                                 ^^^^^^
AttributeError: module 'homeassistant.helpers.config_validation' has no attribute 'int'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1064, in _get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.gpio_integration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/gpio_integration/__init__.py", line 10, in <module>
    from .hub import Hub
  File "/config/custom_components/gpio_integration/hub.py", line 3, in <module>
    from .config_flow import fill_schema_missing_values
  File "/config/custom_components/gpio_integration/config_flow.py", line 42, in <module>
    from .schemas.servo import SERVO_SCHEMA, validate_servo_data
  File "/config/custom_components/gpio_integration/schemas/servo.py", line 23, in <module>
    vol.Optional(CONF_DEFAULT_STATE, default=0): cv.int,
                                                 ^^^^^^
AttributeError: module 'homeassistant.helpers.config_validation' has no attribute 'int'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 344, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1013, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1005, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1064, in _get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.gpio_integration

jatgm1, 2025-05-12:
Logger: homeassistant.loader
Source: loader.py:1053
First occurred: 12:43:22 PM (2 occurrences)
Last logged: 12:43:22 PM

Unexpected exception importing component custom_components.gpio_integration
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/gpio_integration/__init__.py", line 10, in <module>
    from .hub import Hub
  File "/config/custom_components/gpio_integration/hub.py", line 3, in <module>
    from .config_flow import fill_schema_missing_values
  File "/config/custom_components/gpio_integration/config_flow.py", line 42, in <module>
    from .schemas.servo import SERVO_SCHEMA, validate_servo_data
  File "/config/custom_components/gpio_integration/schemas/servo.py", line 23, in <module>
    vol.Optional(CONF_DEFAULT_STATE, default=0): cv.int,
                                                 ^^^^^^
AttributeError: module 'homeassistant.helpers.config_validation' has no attribute 'int'

jatgm1, 2025-05-12:
Logger: homeassistant.util.loop
Source: util/loop.py:77
First occurred: 12:43:22 PM (1 occurrence)
Last logged: 12:43:22 PM

Detected blocking call to import_module with args ('custom_components.gpio_integration',) in /usr/src/homeassistant/homeassistant/loader.py, line 1053: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 706, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 677, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2034, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 171, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 344, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1005, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

rosenkolev, 2025-05-12:
Please, don't post 3-4 messages at the same time, If you forgot something to add there is `edit` possibility.

I have created another version with the fix 2.1.1.

jatgm1, 2025-05-12:
So I tried, but I really don't know how to configure it to get full motion from one side to the other. It is limited to 100 in the number variable but can definitely move further. I can set it to move to one side or the other but it doesn't allow for full movement. The 1-100 variable isn't the percentage of movement out of the range provided but 100 degrees of movement somewhere in the range. I think it just makes the number the angle, and won't go under 0 or over 100. So if it's set to -120° to 120° I can only go from an angle of 0-90. which limits the movement of the servo. Im not sure I'm doing it right... I assumed it would be a percentage of the way through the angles provided? Idk

closed 2025-11-17 by github-actions[bot]